panoptes/POCS #815: related to message api

kmeagle1515, 2019-04-04:
I will be asking doubts related to project idea message based api design and implementation for it's components.

kmeagle1515, 2019-04-04:
@wtgee  @AnthonyHorton  @jamessynge  my proposal is almost done, i have some doubts related to the proposal. Is there a mail id to which i can send my proposal for a review ,so i can make necessary changes if required.

wtgee, 2019-04-04:
Hi @kmeagle1515. All proposals, including drafts, should be submitted formally through the GSoC website. They have a mechanism there to mark it as a draft and for us to leave official feedback on it. Please try to send the proposal via the website and let us know if you are having any problems. Thanks!

kmeagle1515, 2019-04-05:
@wtgee Thanks!
Have done that.

kmeagle1515, 2019-05-07:
@ wtgee can you suggest what went wrong for my project whether it was my proposal or some other things. Also is there any possible way in which panoptes can assign this project to the student who applied (not under gsoc)because i guess we all have studied about the project, spent time understanding panoptes. So please inform if it's possible.

closed 2019-10-10 by kmeagle1515