panoptes/POCS #818: Depreciation at pocs/utils/config.py line 15 : yaml.load_all is unsafe with default loader

SMe12435, 2019-04-07:
In order to tackle is this, we have to add an additional Loader Type parameter.
`Loader = yaml.FullLoader`

There are a couple of choices for the YAML Loader:

- **Base Loader**:Only loads the most basic YAML
- **SafeLoader**: Loads a subset of the YAML language, safely. This is recommended for loading untrusted input.
-  **FullLoader**: Loads the full YAML language. Avoids arbitrary code execution. **This is currently (PyYAML 5.1) the default loader called by yaml.load(input) (after issuing the warning).**

The current situation suggests we should use Full Loader. Any views on using Basic/Safe Loader? @wtgee @jamessynge

wtgee, 2019-04-07:
HI @SMe12435, thanks for looking into this and the other items from #810.

This is actually already solved in #814, I'm just waiting for @jamessynge or @AnthonyHorton to review. :wink:. I ended up using the `full_load` call, which worked well.

However, on another note, I'm currently doing a little bit of surgery on POCS in order to split all of the `pocs.utils` into their own repository. There are times when we want to use the `utils` without requiring a full install of `pocs`. This work is going on over in https://github.com/panoptes/panoptes-utils.

The changes you already did in #816 and #817 will make their way there as will these changes from #814.  I will also start moving Issues from here to there pretty soon, just FYI.

SMe12435, 2019-04-07:
Oh! Sorry for creating a duplicate issue. @wtgee

wtgee, 2019-06-29:
Closed as part of panoptes-utils 0.0.8 release.

closed 2019-06-29 by wtgee