panoptes/POCS #825: Import error while taking individual tests.

kmeagle1515, 2019-04-16:
#793 **Describe the bug**
A clear and concise description of what the bug is.

**To Reproduce**
Steps to reproduce the behavior:
1. Go to '...'
2. Click on '....'
3. Scroll down to '....'
4. See error

**Expected behavior**
A clear and concise description of what you expected to happen.

**Related Issues**
Please link any related Issues if known.

**Screenshots**
[image]
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

kmeagle1515, 2019-04-18:
@wtgee @jamessynge i am getting this import error while running individual tests on my machine. This issue was earlier faced by @ImmortalRabbit  but he solved it by properly setting env variables. 
But i have set those in my etc/profile.d/new.sh file so that is not an issue for me i suppose.

AnthonyHorton, 2019-05-24:
Looks like the problem is due to not running tests from the top level directory of the repo (therefore no invoking conftest) and running `pytest` on a Python file that isn't a test file. Closing as not a bug.

closed 2019-05-24 by AnthonyHorton