panoptes/POCS #828: Installation dependency problems

jamessynge, 2019-05-04:
I'm using scripts/install/install-dependencies.sh to setup a NUC with a fresh install of Ubuntu 18.0.04. Two issues so far:

1. astroscrappy needs a C compiler to install (same is probably true later in the install when we get to astrometry). I added `build-essential` to `apt-packages-list.txt` to work around this.
1. `pip install $POCS/requirements.txt` issues a warning that requests is incompatible with the installed version of urllib3. I wasn't able to see any constraint that explained this, but I was able to fix it with this command: `pip install urllib3>=1.24,<1.25`

I'll send a PR.

wtgee, 2019-06-22:
Should be addressed by #843

wtgee, 2020-02-17:
Will be closed by #951

PR #951 (2020-06-01): Docker into develop 02
Merging the `docker` branch into `develop`.

This PR is effectively `v0.7.0`

## Related Issue
Replaces #950 
Closes #927 
Closes #934 
Closes #948 
Closes #531 
Closes #137
Closes #734 
Closes #960
Closes #828 
Closes #952 
Closes #897 
Closes #809 
Closes #970 
Related to #115 

Relies on https://github.com/panoptes/panoptes-utils/pull/189

## Types of changes

- [x] New feature (non-breaking change which adds functionality)
- [x] Breaking change (fix or feature that would cause existing functionality to change)
closed 2020-06-01 by wtgee