panoptes/POCS #837: Creation of scheduler outside of Observatory class

ImmortalRabbit, 2019-05-29:
**Is your feature request related to a problem? Please describe.**
In order to add dependency injection, creation of scheduler should be done outside of Observatory class.
**Describe the solution you'd like**
I would like to add creation of scheduler outside of Observatory class. In order to do that I am willing to send scheduler into Observatory class as cameras were sent: 
`def __init__(self, cameras=None, scheduler=None, *args, **kwargs):`
or
`def __init__(self, cameras=None, scheduler, *args, **kwargs):`
I am not quiet sure, should I consider situation when scheduler will not be sent to Observatory class. According to my understanding scheduler is just a software and it's not like cameras that can be not attached. 
Then planning to replace A with B:
A:
```
        self.scheduler = None
        self._create_scheduler()
```
B:
```
       self.scheduler = None 
       if scheduler:
            self.logger.info('Creation of the scheduler in the observatory')
            self.scheduler = scheduler
```
**Describe alternatives you've considered**
A clear and concise description of any alternative solutions or features you've considered.

**Additional context**
Add any other context or screenshots about the feature request here.

**Related Issues/PRs**
If there are any related Issues or PRs list them here:
##### Depends on: #836 
##### Required by: 

@wtgee @jamessynge

wtgee, 2019-05-29:
> am not quiet sure, should I consider situation when scheduler will not be sent to Observatory class.

Yes, I think we want to be able to create an Observatory without necessarily specifying a scheduler. As such, you should use the keyword based (i.e. `scheduler=None`) approach in the constructor.

One thing we will need to do is identify the methods inside the `Observatory` that rely on a scheduler and give appropriate warnings (or handle in some other way) if a scheduler is not present.  For example, an observatory cannot `.observe()` without a scheduler so it makes sense to do some checks before allowing that method.

ImmortalRabbit, 2019-05-30:
> One thing we will need to do is identify the methods inside the `Observatory` that rely on a scheduler and give appropriate warnings (or handle in some other way) if a scheduler is not present. For [...]

```
        if not self.scheduler:
            return dict()
```

Or should I raise error than just empty dictionary? Or empty dictionary will cause an error?

jamessynge, 2019-05-31:
The simple approach to dealing with _scheduler being None is simply to raise an exception (e.g. pocs.utils.error.NoObservation) if the observe method is called.

I'd convert .scheduler to ._scheduler, and introduce an @property accessor called scheduler which can then raise the NoObservation error if called when _scheduler isn't set.

ImmortalRabbit, 2019-06-01:
> The simple approach to dealing with _scheduler being None is simply to raise an exception (e.g. pocs.utils.error.NoObservation) if the observe method is called.
>
> I'd convert .scheduler to [...]

In my pull request #838 I added a new error NoScheduler which appears if scheduler is None. Do you think that will be fine?

commit 4476d3b3 (2019-06-11): Added scheduler as arg to Observatory and error NoScheduler (#838)

De-couples the creation of the scheduler from the `Observatory` class.

Closes #836, closes #837, closes #839 

Thanks @ImmortalRabbit, good work!
closed 2019-06-11 by wtgee