panoptes/POCS #846: Hanging camera if autofocus_size is too large

wtgee, 2019-06-16:
**Describe the bug**
If you set the `autofocus_size` larger than the image it will hang until you manually kill. Then you see an AssertionError.

**Expected behavior**
Handle the error gracefully.

```
In [4]: cam.autofocus()
Out[4]: <threading.Event at 0x7f30c4366fd0>

In [5]: exit
Exception in thread Thread-5:
Traceback (most recent call last):
  File "/storage/miniconda3/envs/panoptes/lib/python3.6/threading.py", line 916, in _bootstrap_inner
    self.run()
  File "/storage/miniconda3/envs/panoptes/lib/python3.6/threading.py", line 864, in run
    self._target(*self._args, **self._kwargs)
  File "/storage/panoptes/POCS/pocs/focuser/focuser.py", line 353, in _autofocus
    dark=True)
  File "/storage/panoptes/POCS/pocs/camera/camera.py", line 468, in get_thumbnail
    thumbnail = img_utils.crop_data(image, box_width=thumbnail_size)
  File "/storage/panoptes/POCS/pocs/utils/images/__init__.py", line 57, in crop_data
    box_width, data.shape)
AssertionError: Can't clip data, it's smaller than 5000 ((2532, 3352))
```

wtgee, 2020-02-13:
@AnthonyHorton do we know if this is still an issue?

AnthonyHorton, 2020-02-13:
@wtgee It probably is. I don't remember putting a check in to prevent the issue.

PR #1029 (2020-11-10): Autofocus plots refactor
* Move the autofocus plotting to a separate file.
* Super dooper explicitly close the figure (closes #1028).
* Logging call cleanups.
* Add default param to config file for making autofocus plots (closes #925).
* Make sure thumbnail is not bigger than image (closes #846 ).
* **Change output filenames for consistency**

## Related Issue
Closes #1028 
Closes #925 
Closes #846 

## How Has This Been Tested?
It hasn't!!  We currently don't save the focus images or data in a way that would allow this to be easily reproduced.
commit 132c6aa0 (2020-11-10): Autofocus plots refactor (#1029)

Thanks @danjampro and @AnthonyHorton !

* Autofocus plots refactor:

* Move the autofocus plotting to a separate file.
* Super dooper explicitly close the figure (closes #1028).
* Logging call cleanups.

* * Add default config entry for making autofocus plots (closes #925)

* * Make sure thumbnail is not bigger than image. Closes #846

* * Fix conditional.

* File naming consistency.

* * Consolidating exceptions
* Adding an autofocus tests that doesn't use darks.

* Revert "* Consolidating exceptions"

This reverts commit fd468bdceae42a29dc3b6a09bae011dd96910 [... cut]
closed 2020-11-10 by wtgee