panoptes/POCS #872: pocs_shell doesn't know about a new changes of Observatory class

ImmortalRabbit, 2019-08-09:
**Describe the bug**
Before, we had only create_camera_from_config and pocs_shell used only it, to create Observatory instance. Now it requires create_scheduler_from_config, create_mount_from_config, create_dome_config.

**To Reproduce**
If we will run pocs_shell, it has to show a lot of errors because of wrong Observatory class is done in wrong way. 

**Expected behavior**
A clear and concise description of what you expected to happen.

**Related Issues**
Please link any related Issues if known.

**Screenshots**
If applicable, add screenshots to help explain your problem.

**Additional context**
Add any other context about the problem here.

wtgee, 2019-08-17:
Note that there is work in #859 where this has been fixed which could feasibly be back-ported.

wtgee, 2020-02-16:
This has been cleaned up along the way (although maybe just in #951). Closing for now.

closed 2020-02-16 by wtgee