panoptes/POCS #908: Maintain dependencies in setup.py

wtgee, 2019-10-14:
Move dependency list to `setup.py` and use use the `pip-compile` tool introduced in #907 to generate `requirements.txt`.

**Describe the solution you'd like**
The definitive list of dependencies should be kept in `setup.py` to be compatible with setuptools but there should also exist a `requirements.txt` version for `pip` installing. 

#907 introduced the use of `pip-compile` (from `pip-tools`) for auto-generating the `requirements.txt` file from a `requirements.in` file. We should instead keep the list of dependencies directly in `setup.py` and remove the `requirements.in` file completely.

**Related Issues/PRs**
#907

capsulecorplab, 2019-10-30:
This is also achievable with the pip-tools toolchain; If the dependency list is specified in an `install_requires` in `setup.py`, then you can generate `requirements.txt` from `setup.py`, using `pip-compile --output-file requirements.txt setup.py`.
See [requirements-from-setuppy](https://github.com/jazzband/pip-tools/#requirements-from-setuppy)

This also happens to be the recommended configuration by the PyPA User Guide  :slightly_smiling_face:
https://packaging.python.org/discussions/install-requires-vs-requirements/

wtgee, 2019-10-30:
> This is also achievable with the pip-tools toolchain; If the dependency list is specified in an `install_requires` in `setup.py`, then you can generate `requirements.txt` from `setup.py`, using `pip-compile` without any arguments, or `pip-compile --output-file requirements.txt setup.py` to update.
> See [requirements-from-setuppy](https://github.com/jazzband/pip-tools/#requirements-from-setuppy)
> 
> This also happens to be the recommended configuration by the PyPA User Guide 
> https://packaging.python.org/discussions/install-requires-vs-requirements/

Thanks @capsulecorplab, that's what I did for a different repo, I just need to clean this one up a bit before.  Cheers!

wtgee, 2020-02-13:
This was completed as part of #937

closed 2020-02-13 by wtgee