panoptes/POCS #915: POCS.park() doesn't park bisque mount

AnthonyHorton, 2019-10-22:
**Describe the bug**
When calling `POCS.park()` with Huntsman (which has a `bisque` type `Mount`) there is an unhandled `TheSkyXTimeout` error immediately after sending the command to home the mount, which results in the subsequent mount park command not being sent. Consequently the mount ends up being left at the home position instead of the park position. Seems like POCS is not waiting for the home mount command to complete?

**To Reproduce**
Steps to reproduce the behavior:
1. Create a `POCS` instance with an `Observatory` that has a `Mount` of `bisque` type.
2. Start the `POCS` instance running.
3. Invoke `POCS.park()`.

**Expected behaviour**
The mount first moves to the home position (why?), then to the park position.

**Actual behaviour**
Mounts starts moving to the home position, there is a `TheSkyXTimeout` error, the mount does not park.

wtgee, 2019-10-22:
This probably just needs to implement the `blocking=True` option created in https://github.com/panoptes/POCS/pull/887. The default was left to `False` so there wouldn't be any unexpected changes from what currently existed.

Note that the parameter is currently being passed, just need to build in some kind of check and timeout.

https://github.com/panoptes/POCS/blob/develop/pocs/mount/bisque.py#L206-L223

wtgee, 2019-10-23:
Related: https://github.com/AstroHuntsman/huntsman-pocs/issues/102

wtgee, 2019-10-23:
Not recreating this either.  Seems like with #916 and #917 there was something causing that `TheSkyXTimeout` to occur, which was making each of the commands fail. This was also before we had the odd TSX communication errors. Will leave open for now until we find out more info.