panoptes/POCS #924: Add defocus option to camera Focusers

AnthonyHorton, 2019-10-28:
**Is your feature request related to a problem? Please describe.**
For precision photometry of bright targets it can be advantageous to deliberately defocus the images. While this can be done manually (with the `.move_by()` method of `Focuser` classes) it will be overidden by the autofocus that gets triggered at the start of POCS observations.

**Describe the solution you'd like**
The `Focuser` classes should have an option to apply a set amount of defocus at the end of the autofocus sequence. The amount of defocus that is applied should be configurable on a per camera basis, and it should be possible to enable/disable defocus for specific targets. Ideally it would be possible to scale the amount of defocus, relative to each camera's default defocus, on a per target basis (brighter target may benefit from more defocus, while fainter ones may want less to reduce the read noise penalty).  

**Describe alternatives you've considered**
'Manual' observing to avoid the triggering of autofocus. This has been done once with Huntsman and the large amount of lost automation makes this far from ideal.

**Additional context**

[image]

[image]

wtgee, 2019-10-28:
@AnthonyHorton what does "relative to each camera's default defocus" look like? Number of steps or some kind of percentage?

AnthonyHorton, 2019-10-31:
@wtgee My thinking is that each camera, in its config, has a default amount of defocus in terms of  number of encoder steps, and then targets can apply a multiplier to that default if they want more or less than the default amount of defocus.