panoptes/POCS #926: Making pocs script check if docker containers are already running

zacharyt20, 2019-11-14:
The ```pocs``` bash script that starts the docker containers should check if containers are already running. Currently, it only checks for this when ```bin/pocs up -d``` is used but doesn't when ```bin/pocs up``` is used.
 
**Describe the solution you'd like**
Use an if statement evaluating the output of ```docker ps``` to check if docker containers are already running and exit the script early with an error if they are.

wtgee, 2019-11-14:
Hi @zacharyt20, thanks for the Issue.

> Currently, it only checks for this when bin/pocs up -d is used but doesn't when bin/pocs up is used.

There shouldn't be any difference between the two so I'm curious as to what evidence you saw that there was a difference?

Note that `bin/pocs-cmd` does currently check if running, so the same code could probably work.

In general we should list some things that are desired for the start up, e.g. not start up the weather if not weather station, check for proper config, etc.

zacharyt20, 2019-11-14:
Hi @wtgee , 
I just re-tested the commands again on our unit and they are behaving the same way. Initially I didn't see the first output of ```bin/pocs up -d``` because of all the log messages.

Just by looking at the output that I had missed, ```bin/pocs up``` detects if containers have been started and gives a different text output, but still tries to load the containers. I was just looking in ```bin/pocs-cmd``` and it looks like it is meant to start the docker containers, but I don't see it called in ```bin/pocs```.  Is it called somewhere else and is it even connected to ```bin/pocs```?

I agree that giving the user the ability to specify which containers to start would be a great addition. I know the POCS shell uses a simulator system, would we need to simulate a sensor or simply not have it load? 

Also, I think that determining whether or not docker containers are running should be the first thing that happens in the script.

wtgee, 2019-11-14:
> I was just looking in `bin/pocs-cmd` and it looks like it is meant to start the docker containers, but I don't see it called in `bin/pocs`. Is it called somewhere else and is it even connected to `bin/pocs`?

It doesn't start the containers, it just checks if started or not here:
https://github.com/panoptes/POCS/blob/bfe33157431ec2c72ce7cd6d9193ff7b1884355f/bin/pocs-cmd#L6-L8

The idea with `pocs-cmd` is that it will run a command inside the container from outside the container.  This is what `docker exec` does. So, for instance, if you want to run `jupyter-console` inside the `pocs-shell` container, you call `bin/pocs-cmd jupyter-console` from _outside_ the container, that is, just from the regular command line.

> Also, I think that determining whether or not docker containers are running should be the first thing that happens in the script.

Yes, agreed. I'm not sure how that hasn't come up before.  Want to take a stab at a PR? :wink:

zacharyt20, 2019-11-14:
@wtgee  [Its](https://github.com/zacharyt20/POCS/pull/1) up, but my bash scripting could use some work! (See issue in comments of pocs file on the PR)

wtgee, 2020-02-17:
Will be closed by #951

wtgee, 2020-06-01:
Pulled into #951. Thanks @zacharyt20!

closed 2020-06-01 by wtgee