panoptes/POCS #933: POCS-SHELL: reset_pocs bypasses exit's safety functions

zacharyt20, 2019-12-11:
*In the docker version of POCS, the `pocs-shell` can be exited quickly by using `reset_pocs` before entering `exit`. If the user accidentally types `reset_pocs` while the mount is in the home position, the mount does not automatically park itself, and does not warn the user. This is dangerous as the camera's can be easily damaged if left in the home position during the day. If a unit isn't using an external webcam to monitor the unit, this would likely go unnoticed.*

**Solution**
Warn the user if they try running `reset_pocs` while the mount is not in the park position, and do not execute `reset_pocs`. Allow the user to enter `reset_pocs` again (twice in a row) to reset the pocs instance from anywhere.

**Alternatives**
*Have reset_pocs park the unit before resetting the pocs instance.*
I think this shouldn't be implemented because the purpose of `reset_pocs` is to allow the user to throw away a dysfunctional pocs instance for debugging or troubleshooting purposes. I don't think it would be a good idea to add a possibility that `reset_pocs` fails because the park function throws an error.

**Home and Park Position**
Home position: Head unit is up right and cameras are pointed toward polaris(sky)
Park position: Cameras are facing down towards the ground and the RA axis is horizontal to the ground. It is the safe position for the telescope, where the cameras can't be damaged.

**Reproduce**
Enter the pocs-shell and setup pocs.
Unpark the mount and get it into the home position through `go_home`. (Note: We first found this issue after running the `polar_alignment_test`
Enter `reset_pocs`.

wtgee, 2020-02-17:
Will be closed by #951

wtgee, 2020-06-01:
Closed by #951. Please re-open if necessary after testing.

closed 2020-06-01 by wtgee