panoptes/POCS #941: prepare POCS for a filterwheel mode

lspitler, 2020-02-03:
*Modify POCS to accommodate Filterwheels.*

**Describe the solution you'd like**
Here are the tasks:

- [ ] when stepping through e.g. `target.yaml` each target can have an optional `filter_name` key/name that specifies the desired filter for a target
- [ ] Modify `target.yaml` parser to accommodate the `filter_name` key.
- [ ] Modify Observation class to add an optional filter_name arg to `scheduler.observation.Observation`.
- [ ] In `camera.take_observation`, if `filter_name` was specified and if `filter_wheel_exists=True`, then run `camera.move_to(filter_name)`

- [ ] In `filter_wheel.config`, for each filterwheel require a filter_wheel configure to have a `default_filter_name` taht will be used if someone doesn't speicify `filter_name` in `targets.yaml`
- [ ] Modify the camera initialisation so it sets up a filterwheel.

**Related Issues/PRs**
If there are any related Issues or PRs list them here:
##### Depends on: #940

danjampro, 2020-02-03:
"Modify target.yaml parser to accommodate the filter_name key."

I believe this is done automatically in `pocs.scheduler.scheduler.add_observation`

Also by targets.yaml, we are referring to the value of `fields_file` in the config

AnthonyHorton, 2020-02-03:
> Modify the camera initialisation so it sets up a filterwheel.

Not necessary, that code has been there a while. There just needs to be a filterwheel configuration in the camera config.

wtgee, 2020-05-29:
@danjampro and @AnthonyHorton how is this Issue in relation to the work that has been done?