panoptes/POCS #948: Remove mongo

wtgee, 2020-02-13:
We no longer use mongo for our database storage but instead default to json flat files.  We no longer wish to maintain mongo support as it is just adding extra dependencies that we don't need. 

**Describe the solution you'd like**
Remove mongo from `database.py` and remove all references and usage of mongo. Also strip from the test suite (which will also help with #191).

**Related Issues/PRs**
#191

ac-optimus, 2020-02-16:
hi, @wtgee I would like to work on this issue. I have a small doubt related to the the mention  `Also stip from  the test suit`.  I can not see any reference to `mongo` in any of the `tests`? Can you please guide me on this?

wtgee, 2020-02-16:
> hi, @wtgee I would like to work on this issue. I have a small doubt related to the the mention `Also stip from the test suit`. I can not see any reference to `mongo` in any of the `tests`? Can you please guide me on this?

Hi @ac-optimus , thanks for the interest. This has already been done actually and I just hadn't been good about marking the issue. It's not merged yet but is a part of #951.

I'm happy to suggest some other issues.  Are you looking at a GSoC application?

ac-optimus, 2020-02-16:
@wtgee thank a lot! could you please suggest some issues. I am new to this organisation and would love to contribute. Yes I am also looking at GSOC application, though I was not able to direct myself to the project ideas page. Can you share some information related?

wtgee, 2020-02-16:
> @wtgee thanks a lot! could you please suggest some issues. I am new to this organisation and would love to contribute. Yes I am also looking at GSOC application, though I was not able to direct myself to the project ideas page. Can you share some related information?

I'll try to update the page later today. It's currently just blank! Thanks for the interest. I'll let you know here when updated as well.

wtgee, 2020-02-17:
@ac-optimus actually there is some work related to this in our [`panoptes-utils`](https://github.com/panoptes/panoptes-utils/issues/94) repo if you want to take a stab at it.

Just as an FYI, we don't yet know for sure whether we will be selected as a GSoC project (announced Feb. 20).

ac-optimus, 2020-02-17:
> ~@ac-optimus actually there is some work related to this in our [`panoptes-utils`](https://github.com/panoptes/panoptes-utils/issues/94) repo if you want to take a stab at it.~ Sorry, this will probably be done by the time you read this. :)
> 
> Just as an FYI, we don't yet know for sure whether we will be selected as a GSoC project (announced Feb. 20).

hi @wtgee just saw your message 😅. Can you suggest some other issue that I can try my hands on? 
Yeah, I am aware of that fact but I would still want to contribute!
Thanks

PR #951 (2020-06-01): Docker into develop 02
Merging the `docker` branch into `develop`.

This PR is effectively `v0.7.0`

## Related Issue
Replaces #950 
Closes #927 
Closes #934 
Closes #948 
Closes #531 
Closes #137
Closes #734 
Closes #960
Closes #828 
Closes #952 
Closes #897 
Closes #809 
Closes #970 
Related to #115 

Relies on https://github.com/panoptes/panoptes-utils/pull/189

## Types of changes

- [x] New feature (non-breaking change which adds functionality)
- [x] Breaking change (fix or feature that would cause existing functionality to change)
closed 2020-06-01 by wtgee