panoptes/POCS #954: POCS CLI

wtgee, 2020-02-19:
## Current Setup 
POCS and PEAS are currently controlled via custom shell utilities. These utilities are responsible for:

* Setup/Initialization:
  * POCS:
    * The `POCS` instance, which includes the `Observatory` (which includes the `mount`, `cameras`, `scheduler`, etc.).
  * PEAS:
    * The weather station.
    * The "control board" for power monitoring and relay.
    * The "camera board" for power monitoring and relay of camera box.

* Monitoring:
  * PEAS: via the `last_reading weather`, `last_reading control_board`, etc. This simply shows the most recent entry from the stored reading (a flat json file).

* Control:
  * POCS: via the `run_pocs` command, which blocks until system is shut down. Killing the command (`Ctrl-C`) will power down POCS, which importantly includes making sure the mount is parked.

Both the POCS shell and the PEAS shell are contained shell applications so they must be run in different terminals.  Additionally we usually tail the log files in a separate terminal.

Currently the POCS and PEAS shells hold the reference to the various object instances. The proposed solution would involve creating a daemonized instance of these objects, with the command line interface accessing the daemon. These daemons are running inside docker containers.

### Sample session - current setup

PEAS:
```bash
$ bin/peas-shell 
Welcome to PEAS Shell! Type ? for help
PEAS > load_weather
Loading weather reader endpoint

PEAS > load_control_board
Loading control board sensor

PEAS > start
Starting sensors

PEAS > last_reading weather
********************************************************************************
WEATHER:
{'_id': 'dc643cb9-4f1d-44bb-9248-423132c3dc19',
 'data': {'ambient_temp_C': 7.25,
          'date': '20200219T014309',
          ...snip....
          'wind_speed_KPH': 46.0},
 'date': datetime.datetime(2020, 2, 19, 1, 43, 10, 36000),
 'type': 'weather'}
********************************************************************************
3.9 seconds old

PEAS > last_reading control_board
********************************************************************************
CONTROL_BOARD:
{'_id': '7ba5843a-d5d0-4f2b-8672-121a89278a08',
 'data': {
          'current': {'cameras': 82, 'fan': 22, 'main': 522, 'mount': 128},
          'date': '2020-02-19T01:43:17 GMT',
          'humidity': 1.0,
          ...snip...
  },
 'type': 'control_board'}
********************************************************************************
0.5 seconds old
PEAS > 
```

POCS:
```bash
$ bin/pocs-shell 
Welcome to POCS Shell! Type ? for help
POCS > setup_pocs

POCS > run_pocs
Starting POCS - Press Ctrl-c to interrupt
```

## Proposed solution
A non-blocking POCS CLI that can handle the above tasks as well as allow for future development.

Here are some examples (this does not necessarily represent a true workflow):

```bash
$ pocs weather load

$ pocs weather status
********************************************************************************
WEATHER:
{'_id': '0a6496ea-0955-4edd-91a9-36f9a6ae079f',
 'data': {'ambient_temp_C': 7.25,
          'date': '20200218T102906',
         ...snip...  
         'wind_speed_KPH': 46.0},
 'date': datetime.datetime(2020, 2, 18, 10, 29, 6, 521000),
 'type': 'weather'}
********************************************************************************
58.6 seconds old

$ pocs weather # same as status

$ pocs control_board load
$ pocs control_board  # shows status

$ pocs observatory load
$ pocs observatory status

$ pocs mount unpark
$ pocs mount go_home
$ pocs mount park

$ pocs status

$ pocs setup
$ pocs run
```

**Describe alternatives you've considered**
Just improve the existing shell utilities.

**Additional context**
This needs to be done after the merge of #951.

**Related Issues/PRs**
If there are any related Issues or PRs list them here:
##### Depends on:
#951 

##### Required by:
[`Milestone v1.0.0`](https://github.com/panoptes/POCS/milestone/2)

wtgee, 2020-02-19:
Comments from existing users/builders welcome.

ImmortalRabbit, 2020-02-20:
@wtgee It was mentioned that this project idea requires knowledge of Linux. Will I able to complete this task on mac os x or working on Linux is required? I guess docker will be on Linux but can I complete this task using OSX?

wtgee, 2020-02-20:
> @wtgee It was mentioned that this project idea requires knowledge of Linux. Will I able to complete this task on mac os x or working on Linux is required? I guess docker will be on Linux but can I complete this task using OSX?

Good question @ImmortalRabbit. It could probably be done on a Mac and the Linux should probably just say "shell" or "bash" instead. I'll change the website. Thanks!

We'll find out tomorrow if we're accepted so hopefully we can provide more then!

ImmortalRabbit, 2020-02-20:
> Good question @ImmortalRabbit. It could probably be done on a Mac and the Linux should probably just say "shell" or "bash" instead. I'll change the website. Thanks!
>
> We'll find out tomorrow if we're [...]

Will this task require any kind of coverage tests? The task description didn't mention anything about pytest and nothing about how CLI should be tested. 

P.S. I hope PANOPTES will be accepted. It was a pleasure to work on PANOPTES last summer.

wtgee, 2023-05-23:
Much of this has been done and the rest is probably stale. Closing this particular issue, please open a new issue for further items with the cli.

closed 2023-05-23 by wtgee