panoptes/POCS #966: String formatting errors in exception arguments

AnthonyHorton, 2020-04-16:
**Describe the bug**
In POCS there are a few instance of 

`raise ValueError("Blah, blah, blah, {}!", obj)`

where it should be

`raise ValueError("Blah, blah, blah, {}!".format(obj))`

or

`raise ValueError(f"Blah, blah, blah, {obj}!")`, e.g. 

https://github.com/panoptes/POCS/blob/22532dab81745aa5b805d809b4c494bd5495bc39/pocs/focuser/focuser.py#L242 and https://github.com/panoptes/POCS/blob/22532dab81745aa5b805d809b4c494bd5495bc39/pocs/focuser/focuser.py#L249 

These don't cause problems in POCS itself (beyond some funny looking exceptions) but they do cause problems for the remote cameras in huntsman-pocs, because the Pyro4 network layer ends up trying to serialise `ValueErrors` that contain, in their `args` list, a reference to an object that it can't serialise, resulting in opaque 'circular reference' exceptions from Pyro.

wtgee, 2020-04-16:
Thanks for reporting!

AnthonyHorton, 2020-04-17:
For extra credit someone should craft a regular expression for finding all instances of this! I've found four so far, but I'm not sure that's all of them.

wtgee, 2020-05-29:
Related to this we also have a number of places where we are still passing multiple arguments to the logger, as in:

```py
self.logger.debug("Here's some foo", bar)
```

They are also not really causing problems but it would be good to get them switched to f-strings.

wtgee, 2020-06-01:
I tried to take care of a lot of these in #951.

wtgee, 2020-07-07:
I think I got the rest of these in #983.  The regex doesn't turn up anything else.  Feel free to reopen.

PR #983 (2020-07-07): Minor Updates to logging and dome
* Removing source of random testing coverage.
* Fixing logging calls. Closes #966 (I hope).
closed 2020-07-07 by wtgee