panoptes/POCS #968: Support target tags

wtgee, 2020-05-08:
When adding a target to the scheduler it would be nice to be able to add a series of comma separated tags that would be populated into the metadata db for searching.

Ayushsunny, 2021-05-01:
hello, @wtgee is this issue still persist?

wtgee, 2021-05-10:
Hi @Ayushsunny, thanks for asking and sorry it took me a bit to get back to you.

Yes, this is still an issue and actually one that we were hoping to get done pretty soon. If you want to jump in on this, that would be great.

The basics that need to happen are:

1. Modify the `conf_files/fields/simply.yaml` file to support tags in some fashion.  A "field" currently has two top-level keys, the `field, and the `observation`.  The "tags" would most likely be a yaml list on the `observation` key, so something like:

```
-
  field:
    name: HD 189733
    position: 20h00m43.713s +22d42m39.07s
  observation:
    priority: 105
    exptime: 100
    tags:
        - exoplanet
        - defocus_test
```

2. The tags need to be parsed when adding adding an observation to the scheduler. This happens in the [`panoptes.pocs.scheduler.scheduler.add_observation`](https://github.com/panoptes/POCS/blob/e79ac499c924b0a06eb2944a347f8d9251f6db51/src/panoptes/pocs/scheduler/scheduler.py#L212) method.

3. These tags need to be passed into the FITS headers when writing the file. This is part of the camera and happens in the [`_setup_observation`](https://github.com/panoptes/POCS/blob/e79ac499c924b0a06eb2944a347f8d9251f6db51/src/panoptes/pocs/camera/camera.py#L943) and [`_process_fits`](https://github.com/panoptes/POCS/blob/e79ac499c924b0a06eb2944a347f8d9251f6db51/src/panoptes/pocs/camera/camera.py#L973) methods.

There are likely some more details and I'll try to look at the process a little closer later if needed, but that should be enough to make a reasonable start if interested.

Cheers!  Thanks again for reaching out!

closed 2026-02-14 by wtgee