panoptes/POCS #969: Increase level of detail in logs and exception messages when cameras not ready

AnthonyHorton, 2020-05-14:
The `AbstractCamera` base class includes a properties to check whether a camera is ready to take another exposure, `is_ready`. This property is checked in `take_exposure()`, where an exception will be raised if the camera is not ready. Currently this exception simply states that an exposure was attempted when the camera was not ready, it does not say why the camera was not ready (e.g. exposure already in progress, image sensor temperature not stable, filterwheel or focuser not ready).

The camera readiness checks also do very little logging, the only possible log message in addition to the error message from `take_exposure()` is a warning that the image sensor temperature is unstable, and even this does not include the target temperature, actual temperature and cooling power.

This lack of information is presenting a problem for Huntsman debugging, as camera not ready exceptions are being seen but lack the information to identify the root cause.

**Describe the solution you'd like**
The exception raised by `take_exposure()` if the camera is not ready to take an exposure should include the reasons why the camera was not ready. Additionally the camera logs should include full details of any failed readiness checks.

wtgee, 2020-05-29:
Actually I think that I have added a bunch of this in #950 but that was mostly motivated by the fact that I was experiencing a large number of those failures with the cameras.

Most of what I have added is related to the cooling as I had found that it was mostly a race-condition.  Unfortunately that PR is such a gigantic mess that it's hard to point to the individual change.

I'll update this issue later.

wtgee, 2020-06-01:
Closed by #951

closed 2020-06-01 by wtgee