panoptes/POCS #988: Add support for taking dark exposures using a filterwheel equipped with an opaque filter

AnthonyHorton, 2020-07-18:
**Is your feature request related to a problem? Please describe.**
Some camera types (e.g. CMOS image sensor, interline CCD) lack a internal mechanical shutter or lack the control over the shutter required to take dark exposures. If these cameras are equipped with a filterwheel containing an opaque filter then it is possible to use this block the light in order to take dark exposures, however at present this requires manually instructing the filterwheel to rotate to the opaque filter, taking the dark exposures, then manually instructing the filterwheel to rotate to another filter. This is in contrast to dark exposures taken with a camera's internal shutter, which can be taken by simply passing `dark=true` to the camera's `take_exposure()` or `take_observation()` methods.

**Describe the solution you'd like**
It should be possible to take dark exposures using an opaque filter simply by passing `dark=True` to `take_exposure()`, the same as for dark exposures taken using a camera's internal shutter. Setting `dark=True` should automatically rotate the camera's filterwheel to a named opaque filter. Taking a subsequent exposure with `dark=False` should automatically rotate the camera's filterwheel to the most recently selected non-opaque filter.

**Additional context**
This change has been prompted by the upgrade of the Huntsman Telescope to ZWO CMOS cameras which do not have internal shutters, but do have filterwheels equipped with opaque fllters.

**Related Issues/PRs**
If there are any related Issues or PRs list them here:
##### Depends on:
##### Required by:

wtgee, 2020-07-18:
Thanks @AnthonyHorton. Looking at the PR, I'm wondering if we shouldn't have some kind of expanded `pre-` and `post-take_exposure` mechanism that is easy for the cameras to override.  Ideally we could get this into some kind of decorator or plugin and ideally that would work across all our hardware types (I've been wanting to create a `PanHardwareBase` that inherits from `PanBase` and that all actual hardware inherits from, and just as soon as I have time...).

That might not be possible immediately but would be good to at least discuss what that might look like.

AnthonyHorton, 2020-07-19:
I like the idea, but it will take some thought. One thing that would complicate the implementation is that different ways of taking dark exposures require actions at different points, e.g. using an internal mechanical shutter requires setting some shutter control parameter in the camera driver during the camera's `_start_exposure()` method, while for filterwheel darks we need to call filterwheel methods during the camera's `take_exposure()` method before the call to `_start_exposure()`.

AnthonyHorton, 2020-09-03:
Closed by #989. For now. Will probably revisit this as part of a more general Camera refactorisation.

closed 2020-09-03 by AnthonyHorton