panoptes/POCS #993: Pytest-3 fails with ImportMismatchError

Patronics, 2020-08-14:
**Describe the bug**
Running pytest-3 fails 

**To Reproduce**
Steps to reproduce the behavior:
1. follow [these instructions](https://docs.google.com/document/d/16Z87vPiAsEx9yAN5EwsPXn6x5PMAZWMvJz0p2aF6cE8/edit#heading=h.jo5ov7ebccnb) (substituting python3 for python and pip3 for pip as needed)
run pytest-3 when you reach the pytest step. 

**Expected behavior**
The test succeeds, or at least gets past the first issue 

**Actual behavior**
```
________________________________________________________________________ ERROR collecting src/panoptes/peas/remote_sensors.py ________________________________________________________________________
/usr/lib/python3/dist-packages/py/_path/local.py:721: in pyimport
    raise self.ImportMismatchError(modname, modfile, self)
E   py._path.local.LocalPath.ImportMismatchError: ('panoptes.peas.remote_sensors', '/usr/local/lib/python3.8/dist-packages/panoptes_pocs-0.7.5.post0.dev19+gc6c23c6f.dirty-py3.8.egg/panoptes/peas/remote_sensors.py', local('/var/panoptes/POCS/src/panoptes/peas/remote_sensors.py'))
====================================================================================== 1 error in 0.08 seconds =======================================================================================

```

**Additional context**
Add any other context about the problem here.

This is on a fresh reinstall of Ubuntu 20.04 LTS, on an intel NUC based system.

wtgee, 2020-08-14:
Hey @Patronics, thanks for being pro-active, but you're beating me to the PR. :laughing: 

`panoptes-util v0.2.23` was released just a few hours ago and the PR for POCS is incoming.  If that doesn't clear this up I'll take a look.

Thanks!

wtgee, 2020-10-07:
I'm going to close this as stale. @Patronics please reopen if necessary.

closed 2020-10-07 by wtgee