panoptes/POCS #997: docker build fails in setup-local-environment.sh

danjampro, 2020-08-26:
```
Step 24/28 : RUN mkdir -p "${PANDIR}/scripts" &&     cd "${PANDIR}/scripts" &&     wget $gphoto2_url -O gphoto2-updater.sh &&     chmod +x gphoto2-updater.sh &&     wget -q "${arduino_url}" -O install-arduino-cli.sh &&     sudo BINDIR="/usr/local/bin" /bin/sh install-arduino-cli.sh &&     sudo chown -R "${PANUSER}":"${PANUSER}" "${PANDIR}"
 ---> Running in 0c390e233bbf
--2020-08-25 23:58:52--  https://raw.githubusercontent.com/gonzalo/gphoto2-updater/master/gphoto2-updater.sh
Resolving raw.githubusercontent.com (raw.githubusercontent.com)... 151.101.28.133
Connecting to raw.githubusercontent.com (raw.githubusercontent.com)|151.101.28.133|:443... connected.
HTTP request sent, awaiting response... 200 OK
Length: 6549 (6.4K) [text/plain]
Saving to: ‘gphoto2-updater.sh’

     0K ......                                                100% 4.17M=0.001s

2020-08-25 23:58:52 (4.17 MB/s) - ‘gphoto2-updater.sh’ saved [6549/6549]

We trust you have received the usual lecture from the local System
Administrator. It usually boils down to these three things:

    #1) Respect the privacy of others.
    #2) Think before you type.
    #3) With great power comes great responsibility.

sudo: no tty present and no askpass program specified
The command '/bin/sh -c mkdir -p "${PANDIR}/scripts" &&     cd "${PANDIR}/scripts" &&     wget $gphoto2_url -O gphoto2-updater.sh &&     chmod +x gphoto2-updater.sh &&     wget -q "${arduino_url}" -O install-arduino-cli.sh &&     sudo BINDIR="/usr/local/bin" /bin/sh install-arduino-cli.sh &&     sudo chown -R "${PANUSER}":"${PANUSER}" "${PANDIR}"' returned a non-zero code: 1
```

wtgee, 2020-08-26:
Can you give details on your environment?

wtgee, 2020-11-13:
Closing as stale.

closed 2020-11-13 by wtgee