panoptes/POCS #1002: Camera take_observation() method fails if optional headers argument not given

AnthonyHorton, 2020-09-11:
**Describe the bug**

`pocs.camera.camera.AbstractCamera.take_observation()` raises a `TypeError` if called on the primary camera without setting the `headers` argument. `headers` is supposed to be an optional argument, & defaults to `None`, but `take_exposure()` contains https://github.com/panoptes/POCS/blob/1f74a94a7fc5e81745eb786b8e5789d8c7da8f6c/src/panoptes/pocs/camera/camera.py#L432, which fails if `headers` is not an iterable.

This could be fixed either by making `headers` default to an empty dictionary, or changing `if 'POINTING' in headers:` to `if 'POINTING' in metadata:` The call to `_setup_observation()` 12 lines earlier generates `metadata` and adds the contents of `headers` to it so if 'POINTING' was in `headers` it will also be in `metadata`.

**To Reproduce**
Steps to reproduce the behavior:
1. Create a POCS `Observation` object, and a primary `Camera` object.
1. Call the `Camera`'s `take_observation()` method with the `Observation` as the only argument.
1. `TypeError` is raised.

**Expected behavior**
`take_observation()` can be called without setting `headers` without raising an exception.

wtgee, 2020-10-07:
> This could be fixed either by making `headers` default to an empty dictionary, or changing `if 'POINTING' in headers:` to `if 'POINTING' in metadata:` The call to `_setup_observation()` 12 lines [...]

I decided to go this route in #1009.   It exposed a few other errors and I took the opportunity to clean up a few things.

wtgee, 2022-01-17:
Closed by #1009 (Maybe? It's been done at some point).

closed 2022-01-17 by wtgee